DeMarcoLab/fibsem #30: Autolamella example issue

sroet, 2023-01-27:
Even after the fixes in #29 I still could not run the [autolamella example](https://github.com/DeMarcoLab/fibsem/blob/main/example/autolamella.py) successfully. It errors out in autoscript with:
`There is no patterning application with name: 'autolamella'.` This name is set as default in [this line](https://github.com/DeMarcoLab/fibsem/blob/3110004fa775cb795b006ae5b192141315d7d5a8/fibsem/milling.py#L15). 

I did pip install the wheel for [autolamella](https://github.com/DeMarcoLab/autolamella) in the same env as `fibsem`, is there some extra step to let autoscript recognize it? (feel free to move this issue to the issue tracker of autolamella if you think it is better suited there)

patrickcleeve2, 2023-01-28:
hi @sroet, you are correct this is missing. 

This is an application file for milling. You need to save the following as autolamella.xml and then load it in the XtUI interface (I dont know if it can be loaded through the API?)

Sorry this is an oversight on my end, will update the instructions.

**autolamella.xml**
```xml
<Content Version="Fei ApplicationFile 1.1">

    <!-- Application file for milling silicon (Si) without any gas for Xe PFIB -->

    <!-- Client tag determines where file is displayed to user. 
         Can be XTUI, iFast or a combination separated by comma -->
    <Client xmlns:dt="urn:schemas-microsoft-com:datatypes" dt:dt="string">
	XTUI, iFast
    </Client >

    <!-- Beam, must be Electron or Ion -->
    <Beam xmlns:dt="urn:schemas-microsoft-com:datatypes" dt:dt="string">
	Ion
    </Beam>

    <!-- Pattern filter sets if file will be displayed for a given pattern type.
         Can be set to Line, Circle, Rectangle, CCS, RCS, Bitmap, StreamFile, Polygon
         Can be a combination separated by comma(s) -->
    <PatternType xmlns:dt="urn:schemas-microsoft-com:datatypes" dt:dt="string">
	Line, Circle, Rectangle, Bitmap, Streamfile, Polygon, CCS, RCS
    </PatternType>

    <!-- Gas type, must match name of gas in UI or empty for no gas -->
    <GasType xmlns:dt="urn:schemas-microsoft-com:datatypes" dt:dt="string">
    </GasType>

    <!-- Dwell time per point, in seconds, minimum is 50e-9 (50ns), maximum is 4.5e-3 (4.5 ms) -->
    <DwellTime xmlns:dt="urn:schemas-microsoft-com:datatypes" dt:dt="r8">
        1.000E-6
    </DwellTime>

    <!-- Overlap between points as percentage of the spot diameter. Must be < 100%. -->
    <Overlap xmlns:dt="urn:schemas-microsoft-com:datatypes" dt:dt="r8">
	98
    </Overlap>

    <!-- Volume per Dose, the volume of material that is deposited or removed per dose of the beam in m^3/Coulomb.
         1e-9 m^3/C = 1 um^3/nC. (Si) is 0.27 um^3/nc = 0.27e-9 m^3/C -->
    <VolumePerDose xmlns:dt="urn:schemas-microsoft-com:datatypes" dt:dt="r8">
        0.00000000038
    </VolumePerDose>

    <!-- Blur and Defocus.  
         Blur- the additional beam diameter that is caused by defocussing in m. 
         Note that the total beam diameter is the root of the sum of the 
         squared beam, interaction and blur diameters.
         
         Defocus- Defocus the beam in m to longer or shorter working distance while patterning.
         Positive values will set focus value to a longer working distance.
         Negative values will set focus value to a shorter working distance.

         Only use a Blur or a Defocus value.  Do not use both.
         If no value is to be set for patterning a Blur or Defocus must be set to a 0 -->
    <Blur xmlns:dt="urn:schemas-microsoft-com:datatypes" dt:dt="r8">
        0
    </Blur>

    <!-- Relative Interaction Diameter must always be set to 0. -->
    <RelativeInteractionDiameter xmlns:dt="urn:schemas-microsoft-com:datatypes" dt:dt="r8">
        0
    </RelativeInteractionDiameter>
    
</Content> 

```

sroet, 2023-01-29:
Thanks for the quick response, will report back once I have access to the hardware to test this.

sroet, 2023-01-31:
> will report back once I have access to the hardware to test this.

I have tested this today and got it to work with the following comments:
- We had to manually put the xml inside the pattern applications directory (`program files/FEI/Data/patterning application files` for our system) 
- We then had to restart the server (not just XtUI) before the file got picked up
- Finally we had to change the `Overlap` down to `92` in order to get a non-zero pitch value for our setup (a pitch value of `0` makes Autoscript error out) 

With those comments (and a local variation of the fixes in #29 ) we ran the autolamella example successfully. 
Feel free to close the issue, or keep it open as a reminder to update the instructions :wink:  

Thanks for the help!

patrickcleeve2, 2023-02-02:
Thats great, thanks @sroet for testing + fixing this! I will update the instructions + merge your changes from #29

sroet, 2023-02-02:
quick warning; I did not actually test the changes from #29. 
I tested a version of those changes that I hacked together while working through the example before making #29. #29 should be equivalent (but technically still untested)

patrickcleeve2, 2023-02-02:
OK, gotcha. Will test before I merge, cheers

patrickcleeve2, 2023-02-13:
@sroet  I've added:
- Both autolamella.xml and cryo_Pt_dep.xml to external/thermo/application_files
- Instructions on how to install to INSTALLATION.md
- A more helpful exception to setup_milling if they dont have the application file installed. 

Closing this, cheers again @sroet

closed 2023-02-13 by patrickcleeve2